prusa3d/PrusaSlicer #14192: No selection for Prusa i3 MK3s/+ printer in PrusaSlicer 2.9.0

LAWBear, 2025-02-23:
After several attempts at reloading Prusa image (prusalink-0.8.1.img) to Pi SD, and loading firmware to Prusa i3 MK3S/+ (MK3S_MK3S+_FW_3.14.1_MULTILANG.hex), and reinstalling PrusaSlicer-2.9.0 I have not found any way to add the **Original Prusa i3 MK3S/+** printer.

The first listed available printer is (Original Prusa MK3.5) that offers the following:

**Code*
Enter printer code
Name (optional)
Original Prusa MK3.5
Location (optional)
Enter printer location**

I believe that at some point I added the correct printer (Prusa i3 MK3S/+ to my account.

[image]

Please advise required procedure to enable WiFi management of the PrusaLink.  WiFi often shows valid IP address.

Thank you for your kind support,

L. A. Walter

Jan-Soustruznik, 2025-02-24:
Hello @LAWBear, Thank you for your feedback. If I understand correctly, you're trying to connect your MK3S+ printer to PrusaConnect. Based on your steps, it seems there may have been some mistakes during the installation. I recommend checking the manual for setting up PrusaLink/PrusaConnect: [PrusaLink and PrusaConnect Setup Guide](https://help.prusa3d.com/guide/prusalink-and-prusa-connect-setup-mk3-s-_221744).

If you continue to experience issues, I suggest reaching out to Prusa support. You can find instructions on how to contact our support team here: [Prusa Support](https://help.prusa3d.com/de/article/kundendienst_2287).

Have a nice day

closed 2025-02-24 by Jan-Soustruznik

bombela, 2025-02-27:
No way to add any Prusa printers. I don't get the dialog with the list of 3D printers anymore. See the series of screenshot taken in order.

[image]

[image]

[image]

[image]

[image]

[image]

bombela, 2025-02-27:
If instead of clicking add printer, I click on "Configuration Wizard", it appears to fix the bug. Though I got another bug. Even though I unchecked both "check for software update" and "check for preset update", it still checked them both forcefully afterward. At least the list of Prusa Printer is back when I click "add printer" now.

[image]

[image]

[image]