prusa3d/PrusaSlicer #14194: [FEATURE REQUEST] Make notifications for prerelease builds optional

craigktreasure, 2025-02-23:
**Is your feature request related to a problem? Please describe.**

A minor annoyance when starting PrusaSlicer when a prerelease build becomes available. The notification for the prerelease build is shown every time you start the application whether you're interested in prerelease builds or not.

**Describe the solution you'd like**

Ideally, if I'm interested in prerelease builds, I can opt into receiving a notification for prerelease builds. Otherwise, i'd prefer to only receive notifications for stable builds.

**Describe how it would work**

* If I have not opted into receiving notifications for prerelease builds, prerelease notifications when starting the application should not be shown.
* If I have opted into receiving notifications for prerelease builds, prerelease notifications when starting the application should be shown.
* I there is a stable build newer than my current version, a notification for the newer stable version should be shown.

**Describe alternatives you've considered**

Provide an option to opt out of notifications for prerelease builds.

**Additional context**

None

lukasmatena, 2025-02-24:
It is optional. There is a selector in Preferences. It has been for years.

TerPuc, 2025-02-24:
Hi @craigktreasure, you can change your settings in Configuration > Preference > GUI > Notify about new releases 😊 
I close this issue.

closed 2025-02-24 by TerPuc

craigktreasure, 2025-02-24:
Sure enough. I must have missed it. Thanks.