prusa3d/PrusaSlicer #14195: The use face using SVG tools on Mac crashes

Haidiye00, 2025-02-25:
### Description of the bug

specific operation：

[image]

### Project file & How to reproduce

![Image](https://github.com/user-attachmen

[Slim Can Koozie 1SVG Fail.zip](https://github.com/user-attachments/files/18958409/Slim.Can.Koozie.1SVG.Fail.zip)

ts/assets/fbdbf2b7-ad71-4ec4-bafe-2a6ceb199335)

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9

### Operating system

mac

### Printer model

any

Jony01, 2025-03-10:
Can't reproduce, 3mf contain only tube without svg. 

[image]

closed 2025-03-10 by Jony01

Haidiye00, 2025-03-11:
Do you want to confirm if you tested it on Mac? Windows is normal