prusa3d/PrusaSlicer #14197: "Failed to connect to Prusa Account." Message

briscohumes, 2025-02-25:
Good Morning Y'all,

Here at our organization we are suddely having issues with logging into our Prusa Account and controlling our MK4S using the PrusaSlicer 2.9.0 Application (even the previous version and the 2.9.1 pre-release).  We are able to access the printer without issue via PrusaLink and also the web based version of PrusaConnect, but we have no luck using the PrusaSlicer (first two images below).  Do you have any thoughts as to what the issue could be?

Also during troubleshooting I have updated the firmware on the printer to the latest firmware even though I believe this is a software/network issue.

Thank you,
Brisco Humes

PrusaSlicer Warning Message

[image]

[image]

PrusaLink

[image]

PrusaConnect

[image]

Printer Firmware

[image]

kocikdav, 2025-02-26:
Hello @briscohumes, thank you for reporting. 

Could you please confirm if sending gcode to the printer via PrusaSlicer physical printer PrusaLink works? You can add physical printer via "cog" icon next to Printer in the right side panel. 

And also could you please try running PrusaSlicer 2.9.1-alpha1 from command line as "prusa-slicer-console.exe --loglevel=4"? And paste here full output?

briscohumes, 2025-02-27:
Good Afternoon @kocikdav,

It looks like I am able to send gcode to the printer via PrusaSlicer's PrusaLink!  This should be an ok work around for now, though it would still be nice to be logged in and having a working PrusaConnect.

I am unsure of how to achieve "loglevel=4" but I went to the console in alpha1 with no lock, screenshot below.

[image]

Thank you,
Brisco Humes

WieValdi, 2025-02-28:
I have a similar problem since version 2.9.1-alpha1. I'm using the Mac version. 
I can log into my Prusa account without any problems.  I have no problems logging in when using 2.9.0 either.

Regards Frank

HonP, 2025-03-04:
The same happens to me, but i thought it is because of lost internet connection (when i let computer sleep). So i would like some "remember password" button or password manager integration.

closed 2026-09-29 by prusaslicer-bot[bot]