prusa3d/PrusaSlicer #14200: Issue with PrusaSlicer Avatar photo

rcmaniac25, 2025-02-26:
### Description of the bug

Got an error message popup

### Project file & How to reproduce

1. Opened PrusaSlicer and setup a model
2. Moved on with life...
3. 3 days later, got this error message saying the image was not a jpeg

(this is how the error message looked. Text is near unreadable white text on grey background, dark mode is enabled)
[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1-alpha1

### Operating system

Windows 10

### Printer model

MK4S + MMU3

kocikdav, 2025-02-26:
Hello @rcmaniac25 thanks for reporting. Would you send us the file on the location in the error dialog? (It is possible it will be already gone or rewritten, but its  worth a shot.)

rcmaniac25, 2025-02-26:
@kocikdav This is the file. There's an identical file which is a PNG, but this is the one referenced

[image]

kocikdav, 2025-02-26:
Hmm, that seems to be ok jpg. Is it still giving the "Failed to load" dialog? The avatar is downloaded on login, so logging out and in or starting app instance should download the avatar again.

rcmaniac25, 2025-02-26:
I opened a new instance but didn't get an error. Again, I had PS open for like 3 days without issue before that error suddenly popped up. For the most part, the slicer sat in the background and wasn't being used. So not sure what triggered it.

closed 2026-09-29 by prusaslicer-bot[bot]