prusa3d/PrusaSlicer #14202: Auto Arrange Object Nesting Regression from 2.8.1 to 2.9.0

mjgord, 2025-02-26:
### Description of the bug

In PrusaSlicer 2.8.1, auto arrange with balanced or accurate geometry handling is able to nest objects inside each other. In 2.9.0, these objects are now arranged next to each other instead.

Behavior on 2.8.1:
[image]

Behavior on 2.9.0:
[image]

### Project file & How to reproduce

I've grabbed a few files that demonstrate the issue, to replicate import the 3mf and use auto arrange with balanced geometry handling and 1mm part spacing.

[plating_regression_demo.zip](https://github.com/user-attachments/files/18983680/plating_regression_demo.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0-MacOS-arm64

### Operating system

macOS 15.3.1

### Printer model

Voron Trident

TerPuc, 2025-02-26:
Dear @mjgord, thank you for your report. Internal isssue with number SPE 2706 was created. We will take a look at it.

5jvm0u4, 2025-03-01:
I got the same issue but probably because my title is worded poorly, it didn't caught any attention #13831

mjgord, 2026-09-18:
This issue is still present in 3.0.0-alpha11