prusa3d/PrusaSlicer #14208: Duplicate filename prevents connect from completing upload

redwoodtree, 2025-02-27:
### Description of the bug

When uploading a file through Prusa Slicer connect, if an STL with the same name exists on the USB drive, the upload will stall with no information about why it is stalled.

### Project file & How to reproduce

To duplicate, 

1. Upload an STL to the printer,
2. once it's complete, open a new project, try to upload an STL with the same name, the upload will start but never complete.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Mac OS Sequoia 15.3.1

### Printer model

Mark 4s

neophyl, 2025-02-27:
Probably a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12871

redwoodtree, 2025-02-27:
Yes, this is a duplicate of 12871 , I couldn’t find it when searching, I should have also tried searching for file_name with a space. Anyway, feel free to close this one. Thank you!

neophyl, 2025-02-27:
Your issue, you close it :)  I'm just a fellow user :)

closed 2025-02-27 by redwoodtree