prusa3d/PrusaSlicer #14209: Error 56: "Send to Connect" Random Failures

tnshadle, 2025-02-27:
### Description of the bug

In PrusaSlicer, when I click "Send to Connect", select a printer and attempt to print now, I usually receive one of the two errors shown in the following screenshots. However, it occassionally works. It used to work reliably, and I'm not aware of a software or config change to cause these. Interestingly, if I save the bgcode file to disk and drag the file into Prusa Connect, it uploads and starts the print without problem.

[image]

[image]

### Project file & How to reproduce

Project file attached. Steps above.

[y-motor-holder-R3+3.zip](https://github.com/user-attachments/files/19016380/y-motor-holder-R3%2B3.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11 Pro

### Printer model

MK4S

TerPuc, 2025-02-28:
Dear @tnshadle, thank you for your report. Are you using Physical printer or Prusa Connect tab? Is your Prusa Connect tab responsive? Are you able to browse it? May I also kindly ask you to provide us with your zipped configuration folder? Thank you for your cooperation.

tnshadle, 2025-03-01:
Trying a different method I created physical printers for my 2 MK4S's with a host type of Prusa Connect. When I click the G button to send the print to the printer I get the same 2 errors. This morning it failed 5 times in a row and worked the 6th time. The Prusa Connect tab seems to be working normally.

tnshadle, 2025-03-05:
I can confirm these failures still occur with the latest PrusaSlicer Version 2.9.1-rc1 and Firmware 6.2.2+8853

[image]

tnshadle, 2025-03-06:
This image shows that performing the identical action twice in a row (clicking the G button to Send to Printer) fails the first time and succeeds the second:

[image]

closed 2026-09-29 by prusaslicer-bot[bot]