prusa3d/PrusaSlicer #14213: Speed setting for (Support) Interface where material changes

tuxmobil, 2025-03-01:
**Is your feature request related to a problem? Please describe.**
I use different materials as support interface, like PLA or TPU as interface layer for supports, as you would with soluble
support material. This works quite well, but as the (incompatible) materials don't stick very well together, the interface
often needs some adjusting.
If support material is printed directly on the (bottom) interface layer, it often does not stick since the printing speed for
support material is way too high.

**Describe the solution you'd like**
What  I would like to have is a setting to slow down the printing of just the layer where the material change
happens. That is the first layer on top of the interface material.

Jan-Soustruznik, 2025-03-03:
Hello, @tuxmobil,

I believe this is already possible. Please check:

_Print Settings > Support Material > Top/Bottom Interface Layers_
_Print Settings > Speed > Support Material Interface_

I think this should achieve the behavior you're looking for.

mcavallaro23, 2025-03-30:
It seems to be a problem only with PrusaSlicer; it doesn't happen in ORCA. Let me explain.
I want to print PLA with support interface in PETG. Immediately after printing the last layer of the PETG interface, the first layer of PLA is printed too fast over the PETG layer (at the same speed as the infill speed of the rest of the piece, let's say 100mm/s). This causes the PLA to detach immediately. However, in ORCA SLICER, by default, the infill speed of the first PLA layer (right above the last PETG interface layer) is much slower (about 20mm/s). I don't see any option in PSLICER to reduce that speed.

One more thing. In ORCA, before printing the first PETG layer (that is, between the support and the interface), it prints a final smooth layer of PLA. However, in PSLICER, the first PETG layer is printed directly on top of the support (which is not a smooth surface but has gaps), forcing me to reduce the support spacing to improve the adhesion of the interface.

I think I was clear, but if images or files are needed, let me know.

mcavallaro23, 2025-03-30:
> It seems to be a problem only with PrusaSlicer; it doesn't happen in ORCA. Let me explain. I want to print PLA with support interface in PETG. Immediately after printing the last layer of the PETG [...]

A solution (other than using ORCA) would be the following:  

1) Add a box-type modifier that covers the entire piece and has a height equal to your layer height.
2) Identify the height at which the interface printing ends and the main material resumes (i.e., the first layer printed with the main material after finishing the interface).  
3) Place the modifier at the height of the layer just before that layer (if the layer is at a height of 5.4mm and you print with a 0.2mm layer height, place the modifier at 5.2mm).  
4) Now, set the speed for that modifier as follows:  
   - **Perimeters:** 10mm/s  
   - **External Perimeters:** 10mm/s  
   - **Infill:** 10mm/s  
   - **Solid Infill:** 10mm/s  
   - **Small Perimeters:** 3mm/s  

All of this will ensure that the speed of the first layer after the interface is printed at a low and safe speed.  

Last but not least, add a G-code temperature change at the mentioned layer height (5.4mm in this example) using:  
M104 SXXX (where **XXX** is the desired temperature).  

For example, if the main material is PETG and you're printing at 250°C, lower the temperature to around **230°C** with:  
M104 S230  

Then, on the next layer (5.6mm), return to the normal temperature with:  
M104 S250 (for 250°C, for example).  

This helps prevent oozing or clogging due to the low print speed on that specific layer.

closed 2026-09-29 by prusaslicer-bot[bot]