prusa3d/PrusaSlicer #14217: Purge tower treated as object while arranging (Not desired)

5jvm0u4, 2025-03-03:
### Description of the bug

Purge tower treated as object while arranging, resulted in wierd placement of EVERYTHING while the tower got fixed in the center for seemingly no reason.
There's already 2 issues (to my knowledge) reporting this exact bug but both had picked up no attentions, since I also got the same experience with other issue (#13831), I think it's the problem of poorly worded title. I'm trying a different way to discribe this bug and link both of the issues here #14115 #13703 

Desired solution:
Ignore purge tower entirely while performing auto-arrange, let the user deside where to put it.

Alternative solution:
Let the delete button work on the purge tower, as if it's the same as disabling it.

What actually happened:
[image]
What it should be:
[image]

### Project file & How to reproduce

Open PrusaSlicer 2.9.0, turn on purge tower, add some objects and hit auto arrange.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

24H2

### Printer model

Original Prusa i3 MK4S with MMU3

halofx, 2025-03-03:
I agree this can be an annoying behaviour. Perhaps we could get a checkbox or toggle in the Arrange Options dialog to "Ignore Purge Tower" when arranging. Much like the current option to "Enable Rotations".

5jvm0u4, 2025-03-07:
Another alternative solution: Fix the purge tower in-place and arrange everything else as if selecting everything else and perform a shift+A, but instead of arranging it around the tower, arrange it aroung the bed center.

Jan-Soustruznik, 2025-04-09:
closed as duplicate of [#14115](https://github.com/prusa3d/PrusaSlicer/issues/14115)

closed 2025-04-09 by Jan-Soustruznik