prusa3d/PrusaSlicer #14222: XL 5T, Slicer 2.9.0 & 2.9.1, XY drift along Z (solved when use slicer 2.7.4)

Bourin01, 2025-03-04:
### Description of the bug

having XY drift resulting in blended parts, 
Problem observed on 2 different printers using FW 6.2.0 and previous unidentified release (haven't updated FW for almost a year before i got this issue)

My issue real look like the note on the last FW release (6.2.1) :
**"A recently discovered numeric error accumulation bug may cause layer shifts, particularly on prints using gyroid infill.
A fix is currently being tested and is expected to be released in an upcoming firmware update."**

But, this only happen when i use Slicer 2.9.1 or 2.9.0, and rolling back to 2.7.4 complete solved the issue (with same settings)
So it look like this is more related to Slicer than the FW
Normal print generated by Slicer 2.7.4:
[image]

[image]

[image]

Drifting prints from Slicer 2.9.1 & 2.9.0 (each file drift at the exact same spot and the same way on both printers) :

[image]

[image]

[image]

Ps : both printers were massivelly trouble shouted since we got the problem, now those are super fine on HW side, and differences between 2.7.4 and 2.9.X were done one after the other (so Only the slicer version changed, as print setting are ported, also note that i also tested without IS and get the same drifting results)

### Project file & How to reproduce

find a 3mf with the settings used in 2.9.0 & 2.9.1 : 

[FailTest-2.9.1.zip](https://github.com/user-attachments/files/19069193/FailTest-2.9.1.zip)

(can't share the exact same part, but drift is also happening on this one, settings used is same as the one we use)

The same 3mf settings used in 2.7.4 :

[FailTest-2.7.4.zip](https://github.com/user-attachments/files/19069554/FailTest-2.7.4.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1 Beta 1 & 2.9.0 official release

### Operating system

Windows 10

### Printer model

XL 5T

Bourin01, 2025-03-04:
Note that this drift issue can damage the docks placements 
(the printer drift more and more until the dock placement is so far away that it violently reline by force at a point, resulting in a massive layer shift to compensate the drift, if you get this kind of problem, just don't inssist to much or you'll have to redo dock calib and maybe have to change linement parts)
The last attempt of using 2.9.1 just after making perfect prints with 2.7.4 :
(bended from start)
[image]

Bourin01, 2025-03-04:
[image]
If you speak about this setting, 
I already tested with and without and got drift

Bourin01, 2025-03-04:
[image]

Bourin01, 2025-03-04:
[image]
I confirm that the 4 parts on last photo have ramming enabled (can't share the 3mf) at various extruded volume and speeds.

this one have none :

[image]

Bourin01, 2025-03-04:
Just sent this one to the printer

[FailTest-v2-2.9.1.zip](https://github.com/user-attachments/files/19071559/FailTest-v2-2.9.1.zip)

just stopped the test without ramming and got drift :

[image]

Bourin01, 2025-03-04:
[image]
Ran the test file for couple hours, I look fine,
So I send the big part in test for the night,
Will see tomorrow if it drifted or not....
PS : the files failed with ramming 'on' were on super low extruded volume settings (0.1 to 1 cubic mm) could explain things....

Bourin01, 2025-03-04:
[image]
Ramming settings under test for Slicer 2.9.1

hejllukas, 2025-03-05:
Hi, the reason why layer shifts don't occur with PrusaSlicer 2.7.4 is that the issue in firmware 6.2.0 depends on a specific sequence of movements with certain parameters (speed, acceleration, and jerk).

Between PrusaSlicer 2.7.4 and 2.9.X, there have been many changes (e.g., seam placement, profiles), which means the generated g-code differs. As a result, the conditions triggering the firmware issue may not be present in g-code from 2.7.4, but this is purely coincidental. There are definitely cases where layer shifts could still happen with PrusaSlicer 2.7.4 and not with 2.9.X.

The best solution is to update the firmware to 6.2.2, where the issue has been fixed, or downgrade to 6.1.3. I'm pretty sure that the g-code from PrusaSlicer 2.9.X will not layer shifts with either of these firmware versions.

Bourin01, 2025-03-05:
@SachCZ @hejllukas ok, I printed the full 'problematic' part using 2.9.1 and the 'normal' ramming setting,
Result is about perfect 👌 

[image]

[image]

So to me last test is : 
Upgrade to fw 6.2.2
And run the same print WITHOUT ramming
And see how it goes

Bourin01, 2025-03-05:
@SachCZ @hejllukas @Jan-Soustruznik just stoped the the test using fw 6.2.2 with slicer 2.9.1 beta 1 and ramming disabled, cause it was doing exact same drift seen on previous FW (made painful sounds on docks due to the huge drift) : 

[image]

(Drifting from start, exactly same as 6.2.1 results...)

Bourin01, 2025-03-05:
[image]
Maybe this photo is better

SachCZ, 2025-03-08:
Hello, not really, the problem is very hard to debug and there might still be firmware issues involved. We will see what we can do.

closed 2026-09-29 by prusaslicer-bot[bot]