prusa3d/PrusaSlicer #14224: Wipe tower location no longer available in print settings

NielschB, 2025-03-04:
### Description of the bug

The wipe tower location on the bed seems no longer present in the print settings

[image]

Screenshot from 2.8.1

[image]

### Project file & How to reproduce

Print settings > Multiple extruders > Wipe tower 
Wipetower X/Y location is no longer present

No specific project required

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1_RC1 Windows

### Operating system

Windows 11 pro 24H2

### Printer model

Custom model based on mk3 + MMU

lukasmatena, 2025-03-04:
Wipe tower position and rotation should never have been in Print Settings to start with. They were removed on purpose in 2.9.0. What do you need them for?

NielschB, 2025-03-04:
When I start Prusaslicer, the wipetower is now by default at the center of the bed. On object insert, these are also placed by default on bed center and "disapear" in the wipetower, so I have to move the wipetower to the side every time. With the XY position I had the wipetower placed by default at the bed edge, for me more convenient location. One other solution could be to hide the wipetower until two or more filaments are selected as for just one filament the wipetower is not required anyway.

svake, 2025-03-07:
> Wipe tower position and rotation should never have been in Print Settings to start with. They were removed on purpose in 2.9.0. What do you need them for?

So far I've usually been happy with Prusaslicer's improvements. This is the first time I disagree with it. Why does it hurt to have the option to change the wipe tower's rotation? On the XL it's always rotated the wrong way by default:

[image]

Since my first print I had the wipe tower rotated 90°, closest and centered to the toolheads on the XL. It was just the logical choice. When I'm not using all the toolheads I move it more to the sides of the toolheads that are used most during a print, but the rotation has always been parallel to the toolheads so it doesn't take up much space. Without rotation it's just in the way. It's not a big deal, but removing that option doesn't improve anything.

murk-sy, 2025-03-08:
Related issue but on the opposite side (though still not fully implemented):
- https://github.com/prusa3d/PrusaSlicer/issues/13442

I do think default position should be near the back corner where the tools are, possibly rotated 90°, see #14241

lukasmatena, 2025-03-10:
Thanks everyone for the input. Just to clarify why the change was made. The wipe tower in Print Settings leads to two major problems. One is described in the other thread that @murk-sy just referenced:

> Any time I slightly touch the wipe tower it shows print settings have been modified, which makes me lose track if something was actually changed

That's basically why I said that it should never have been there.

The other issue is related to multiple beds, which were implemented in 2.9.0. Right now, Print Settings are shared across all bed, while the position of the wipe tower can be different on each bed. That's why it was not possible to not just let it be. We decided to unbound the position of the wipe tower from configuration tabs and it is now positioned just like any other object - both moving and rotation is accessible in the scene.

I understand that this creates the issue with "defaults". This is actually more general issue, which shows up with any "project-specific" configuration, like the purging volumes matrix. We will think about how to solve this cleanly and hopefully improve it the next release.

svake, 2025-03-10:
Thanks for the information. So for the moment I will go on with the 2.8.1 version where I could still rotate the wipetower.

lukasmatena, 2025-03-10:
> I will go on with the 2.8.1 version where I could still rotate the wipetower

You CAN rotate the wipe tower in 2.9.0. You just do it in the scene now, and the default (when opening a new project) is 0 degrees.

dwineman, 2025-03-15:
Can the wipe tower's x/y coordinates and rotation be made editable in the right pane under the object tree when selected, like they are for all other objects? The Z position, X/Y rotation, and scale fields would all have to be disabled, but that seems like the best place for these settings.

@lukasmatena The reason I need this is that sometimes it's necessary to make very fine adjustments in order to fit the tower on the bed without overlapping a large object. This would also make it easier to copy the coordinates onto another bed or project.

murk-sy, 2025-11-25:
You can use sidebar tools or hotkey R to enable rotation. Can't enter exact numbers but considering the wipe tower is fluidly resized based on print settings, I personally find it perfectly workable

[image]

closed 2026-09-29 by prusaslicer-bot[bot]

Alienbliep, 2026-10-26:
Well i still want it on a different spot and not in the middle of my print everytime i start prusaslicer. It is freaking anoying.

[image]