prusa3d/PrusaSlicer #14225: NO ZHop after tool change

Mikmikbt, 2025-03-04:
### Description of the bug

Hi all. I have a problem with changing filament when I do a reverse print.In practice, after having made the change, the nozzle returns to the piece (first layer therefore) and without doing any lifting it goes to make the purge tower. This makes the nozzle slide on the piece In your opinion, is there a way to have a lift done before moving to the blowdown tower? Is it a  bug? Thank you very much

### Project file & How to reproduce

[REVERSE_TEST.zip](https://github.com/user-attachments/files/19074111/REVERSE_TEST.zip)

First layer
Go to  line 2756 (last before tool change)

After tool change, the nozzle goes to the last coordinate than go to purge tower without any lifting.
I need a zhop here

### Checklist of files included above

- [x] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1 beta

### Operating system

Windows

### Printer model

Ender 3 V2 . Srite Pro extruder

SachCZ, 2025-03-04:
Hello, it is assumed that the tool change will do a z lift on its own. If it does not you can simply modify your toolchange gcode:

```
G1 Z{layer_z + 0.2}
T{next_extruder} ; Cambia all'estrusore successivo
;START_CUSTOM_TOOL_CHANGE LAYER
[layer_z]/[layer_num]
{if layer_num >= 0}
M600 ; Pausa per il cambio colore
{endif}
;ENF_CUSTOM_TOOL_CHANGE
```

SachCZ, 2025-03-20:
Hello, I sadly struggle to understand what your problem is. The toolhead is not returning to the piece after the tool change. Instead, it goes directly to wipe tower.

See this gcode:

[image]

X195 Y40 are the wipe tower coordinates.

If your toolchange needs lift, then just add it to the toolchange gcode, no matter before or after the toolchange, you can do both.

closed 2026-09-29 by prusaslicer-bot[bot]