prusa3d/PrusaSlicer #14226: Sync Filament Color with XL and/or MMU3

tneff03, 2025-03-04:
It would be great if the "Set as Current" button in Prusa Connect would synchronize the color of filament as well. 

Our Prusa XL is used by 20 different people, many of whom start prints remotely, with no good way to know what color is in each tool head.

We would like it if the XL would ask what color of filament is being loaded, and it could then be synchronized with PrusaSlicer. 

This would be similar to BambuLab's AMS or ANYCUBIC's ACE Pro, where you select the material and color for each filament as you load it. Then inside BambuStudio, there is a button to "Synchronize filament list from AMS".

So the workflow would be the same as loading a different material, where the XL would ask what material is being loaded, then on the subsequent screen it would ask what color, perhaps with icons similar to the BambuLab AMS.

Then on Prusa Slicer's side, the "Set as Current" button would pull in the type of material as well as the color that was selected.

This would greatly improve our workflow.

See BambuLab's wiki for loading the AMS for pictures: https://wiki.bambulab.com/en/x1/manual/ams-setup-and-filament-loading

gunnaraas, 2025-09-13:
+1, even for printers only used by one or two people, this feature would be a major usability improvement for multi material printing.

hagre, 2026-02-14:
Don't forget to add the filament id for spoolman and/or OpenPrintTag

closed 2026-09-29 by prusaslicer-bot[bot]