prusa3d/PrusaSlicer #14229: Ran of out stack space while trying to match regular expression

michaelaswood1980, 2025-03-05:
### Description of the bug

I am trying to implement a regular expression in PrusaSlicer as part of some customiseable g-code for RatRigs RatOS.
I am following the instructions here: https://os.ratrig.com/docs/slicers/#change-settings-based-on-feature-optional that requires me to add the following under the: Print Settings - Output Options - Other.

[image]

When I then go to slice the model with these settings I get the following error

[image]

### Project file & How to reproduce

The only steps needed are to try and slice the model.

[screen_m4.zip](https://github.com/user-attachments/files/19094696/screen_m4.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1-rc1

### Operating system

Windows 11

### Printer model

Ratrig V-Minion, Prusa XL

earlgreyaroma, 2025-03-22:
I have the exact same problem. But it only happened after I created a new preset (copy of my main one) and set my print setting to "spiral vase". My other presets work fine.

earlgreyaroma, 2025-03-22:
After a bit of testing, it seems that, if I go below a certain combination of infill and perimeters on a print. It throws that error. This combination of settings depends on the size of the print.
i.e. for smaller prints i can get away with less infill and less perimeters

For now I think it's best to remove this G-Code substitution.

smooreace, 2025-05-10:
Same trouble here... also on a Ratrig. Vcore4 500 IDEX

closed 2026-09-29 by prusaslicer-bot[bot]