prusa3d/PrusaSlicer #14231: Move location of bed number in default naming for multiple bed export.

Shinomorisama, 2025-03-05:
**Is your feature request related to a problem? Please describe.**
When adding multiple beds to the printers memory it takes a while to wait to scroll all the way to the end to find the correct file to print.

**Describe the solution you'd like**
Move the bed descriptor to right after the file name in the automatic file naming.
current:{input_filename_base}_{layer_height}mm_{printing_filament_types}_{printer_model}_{print_time}_**{bed}**.gcode
request:{input_filename_base}_**{bed}**_{layer_height}mm_{printing_filament_types}_{printer_model}_{print_time}.gcode

neophyl, 2025-03-05:
You can alter the naming order however you like.  Just edit your profile and save as a custom one.  
Print Settings>Output Options>Output Filename format.  Its a Red/Expert level setting so make sure you are in expert mode.

[image]

Shinomorisama, 2025-03-05:
That is where I got the current implementation from, but I made up the variable(and couldn't do what you suggested without knowing it) for the bed since I couldn't find what that would be in the documentation and don't know git/coding well enough to dig through the code and find it myself.   But the request is also to change the default behavior so I don't have to do custom versions of every prusa printer variant I am slicing for(currently 16 hoping to expand to 24+ soon) .

MrTrick, 2025-05-17:
I came here to ask for this same QoL tweak.

But I am confused.
- Is there actually a specific bed number parameter that can be added into the output filename format?
- If so, is it active (eg 'bed1') even when only a single build plate worth of parts is present?
- The default name settings, eg `{input_filename_base}_0.4n_{layer_height}mm_{printing_filament_types}_{printer_model}_{print_time}.gcode` don't mention bed number at all - so how does the software know to add it? (and where)

Suggestion - If no bed number is specified in the setting, perhaps the default behaviour could be changed 
- Instead of: "append `_{bed}` to the entire filename"
- Do: "append `_{bed}` to the `input_filename_base` value.

RPhilbo, 2025-05-18:
Probable the same request:
#14041

Shinomorisama, 2025-05-19:
> Probable the same request: [#14041](https://github.com/prusa3d/PrusaSlicer/issues/14041)

Related but different.  If you output all beds it does label them.   That one is request to label if not outputting all the build plates.  Which is a separate valid issue.   This is a request to move where the label is by default and I suppose also to document how to change where it goes ourselves.

Shinomorisama, 2025-05-19:
> I came here to ask for this same QoL tweak.
>
> But I am confused.
>
> * Is there actually a specific bed number parameter that can be added into the output filename format?
> * If so, is it active (eg [...]

I am sure there is a parameter that could be added but I could not find it in the documentation.  The example I provided I just made up the parameter to demonstrate current and desired behavior. 

I don't know if it would add it on a single bed in that case since I couldn't find the actual parameter to test with but that wouldn't surprise me.

The software only adds it when you export multiple beds at one time and it goes at the end.  

Your examples are exactly the current behavior and the what I would like the behavior to be.

Being able to edit it would be a good start but I would really like the default behavior to be changed.  I currently have profiles for Mk3s, Mk4,Mk4s, Mini+, 2head XL. All with 4 nozzle sizes and a core one sitting in the cart waiting on funds to hit buy.  That's over 20 profiles and soon to be more.   I don't want to have to do custom ones for all of those especially when I suspect everyone else will also find it annoying to have to wait for it to scroll to the end to find the right print once they get to using that feature.

closed 2026-09-29 by prusaslicer-bot[bot]