prusa3d/PrusaSlicer #14234: custom start G-code not included in exported G-code file

hitmusic100, 2025-03-06:
Works fine in V2.7.0, fails in V2.8.0, fails in V2.9.0

TerPuc, 2025-03-06:
Dear @hitmusic100, unfortunately we cannot reproduce this issue. May I kindly ask you to provide more info about your case? 3mf files, printsceens etc. Thanks!

hitmusic100, 2025-03-06:
completely uninstalled and removed the relevant AppData (anything prusaslicer in Appdata).  It's really factory - so here is the overview:

[image]

I added some custom g-code to the start gcode:

[image]

I then generated the Gcode from within prusaslicer:

[image]

As you can see the entries I made in the start gcode box are not included

Have now saved the .3mf and attach it together with gcode file genereted by prusaslicer

[calendar v6.zip](https://github.com/user-attachments/files/19114961/calendar.v6.zip)

As I say - it works in v 2.4.0

neophyl, 2025-03-06:
The problem is you have that block of code inside a If<>Else <>Endif Statement.  So whats being added is the 

`{else}
G1 Z0.2 F720
G1 Y-3 F1000 ; go outside print area
G92 E0
G1 X60 E9 F1000 ; intro line
G1 X100 E12.5 F1000 ; intro line
{endif}`

It will only add the section you indicated if this evaluates 
{if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/}
Which its not so its adding the contents of the Else.

hitmusic100, 2025-03-07:
Hi @neophyl 
Aaah the penny has dropped - thank you so much for your help! 

Thanks also, @TerPuc 

rgds

closed 2025-03-07 by hitmusic100