prusa3d/PrusaSlicer #14237: [Feature Request] Purge Tower Speed Settings

zjohnson16, 2025-03-06:
On my MK4S I'm currently printing a lot of silk filament which requires lower perimeter speeds for good surface finish. However, the purge tower could be printed at a higher speed since it is purely structural, saving a lot of print time.

I think it would be a nice QOL feature to see an independent speed setting for the purge tower, and perhaps even two different speed settings, one for the exterior of the purge tower and one for the interior.

And yes, I am aware of the gcode-substitution hack that has floated around, but it would be nice to see a feature with greater functionality that is more firmly integrated into the slicer.

Braintoe, 2025-03-13:
I second this one, but with the opposite intention: Prusaslicer freqently prints the Wipe tower too fast for my needs. It would be great to lower the speed as needed.

Braintoe, 2026-09-06:
Still interested in that (see last comment) therefore adding this comment on order to avoid closing the issue.