prusa3d/PrusaSlicer #14238: Non-functional extruder crash-detection

pylorak, 2025-03-07:
### Description of the bug

Toolhead crash detection is confused about objects names if objects are reordered and renamed before reopening project.

PS says the toolhead will crash into object 8 while printing object 2. This is obviously wrong as these are very far apart objects, not to mention object 8 is not even printed yet while printing object 2. If the project is saved, closed and reopened, the reported names become correct.

### Project file & How to reproduce

Because the issue goes away if the project is saved and reopened, after-reorder.3mf is provided only for reference but cannot be used to reproduce the issue. You have to start from before_reorder.3mf.

To successfully reproduce the issue, the following steps are necessary:

1. Open before_reorder.3mf. At this point print order is like in first picture.
2. (Optionally slice the project and no issues are reported.)
3. Without actually moving any objects on the bed, reorder them by moving their names in the list of objects on the right.
4. Edit the object names in the list to reflect their print order. After this step the print order should look like the second picture.
5. Slice the project. A toolhead crash is reported between the wrong objects.
(By the way, even with the correctly reported names, there is no actual crash even in this case. So crash warning is a false positive.)

Project files: [bugreport.zip](https://github.com/user-attachments/files/19117869/bugreport.zip)

Before reorder:
[image]

After reorder:
[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.9.1-rc1-win64

### Operating system

Windwos 11

### Printer model

Voron Trident 300mm

lukasmatena, 2025-03-07:
Thanks, reproduced. We will fix it hopefully before 2.9.1 final.
SPE-2710

Jan-Soustruznik, 2025-03-10:
Closed as fixed in latest version of [PS 2.9.1-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-rc2 )

closed 2025-03-10 by Jan-Soustruznik