prusa3d/PrusaSlicer #14241: Wipe tower rotation angle is missing

svake, 2025-03-07:
Since installing Prusaslicer 2.9.0 there is no wipe tower angle available. I just tested 2.9.1 beta and there the issue still remains.

[image]

In 2.8.1 it was still there:

[image]

u89djt, 2025-03-07:
According to the devs, the position and rotation should never have been in the settings https://github.com/prusa3d/PrusaSlicer/issues/14224#issuecomment-2698289699
I guess the idea is that you just rotate it as you would any object. You can set your defaults in the relevant .ini
[image]

lukasmatena, 2025-03-10:
Duplicate of #14224. Closing.

closed 2025-03-10 by lukasmatena