prusa3d/PrusaSlicer #14245: [Feature request] Retract/unretract inside part boundary

richfelker, 2025-03-07:
I've just switched to PrusaSlicer from Cura, and the one one quality regression I've encountered is that PrusaSlicer performs unretraction for starting the outer perimeter with the nozzle already on the perimeter, creating a bulge at the seam. This is independent of seam gap setting; it's not caused by overlap of the start and end of the perimeter, just by flow behavior on restart.

The way Cura avoids this is by performing the unretract for starting an outer perimeter with the nozzle over where an inner perimeter would/could be (it may already be there or not depending on perimeter order), then moving unretracted to the actual start position. I'm not really convinced this is ideal; it may be better to actually alter the extrusion paths so that the outer perimeter starts/ends inset. But some variant on the concept of "avoid unretract on the outer perimeter" seems like the right thing to do.

The Cura behavior also adds a final travel inward before unretract, which could be useful on machines with slow retraction that ooze while retracting, but this seems a lot less important.

Of course none of this is possible for thin parts that don't admit an inner perimeter, but it makes a very big difference to dimensial accuracy/part fit at the seams when it is possible.

Mike8-I, 2025-03-08:
Thatcher World rallye be great!👍

richfelker, 2025-03-10:
Thinking more about it, a smarter less hackish way to do this (vs what Cura does) would probably be a feature you could call "inset seam". It would be roughly equivalent to making a zero width (or epsilon-thin) cut inward at the seam location perpendicular to the outer perimeter, with length roughly the external perimeter line width, and recessing the actual seam inside that cut.

richfelker, 2026-09-02:
This request still stands. Pinging to quiet the stalebot.