prusa3d/PrusaSlicer #14248: Add Conic Supports like Ultimaker Cura

raffa1799, 2025-03-07:
prusa snug support on platform only

[image]

prusa snug support  everywhere

[image]

cura normal supports  on platform only with conic supports activated

[image]

[image]

pylorak, 2025-03-08:
Prusa has "Organic" support style for the same use case. Did you not see this feature, or do Conic supports have an advantage over Organic?

raffa1799, 2025-03-08:
I know very well that on Prusa Slice you can use tree supports, but I noticed that sometimes all the retractions generated during the creation of the tree support lengthen the printing time and also the slicing time, when I print large components and I have little space left on the plate, if I set the base of the tree support in such a way as to be able to generate gcode that does not come out of the plate, in the end the tree support is too unstable to support high surfaces, with the conical snug supports I could have a thin but long, continuous base that  rests only on the plate

kubispe1, 2025-03-10:
Interesting. Honestly, I have never played with this type of support there, so I don't know how they perform.

Tupson444, 2026-09-19:
still relevant

Jan-Soustruznik, 2026-09-30:
Hello, @Tupson444 , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about. Thank you, and have a nice day.