prusa3d/PrusaSlicer #14250: Text input frequently unreliable, seems to escape text box

Phallon, 2025-03-08:
### Description of the bug

I frequently and randomly run into issues trying to edit text fields for object attributes. Instead of typing text or deleting text in the text box I have highlighted the keystrokes hit Prusaslicer as general key presses doing other things in the UI. This could be a display bug where it's not properly providing feedback you are in text edit mode in the text box or keystrokes being received by the UI instead of the textbox.

My most recent use case:
I'm trying to emboss a logo from a couple SVGs onto a model. I can emboss the SVGs fine, but when I go to edit the details of the SVG, particularly the Depth field, the default + & - buttons are to granular & default to 1mm steps where I need .2 or .4mm. On the first color in the logo I was able to select the numbers & use the keyboard to edit them. On the 2nd color in the logo whenever you try to modify the values it does not type over the existing numbers & hitting back space or delete just deletes the object out of the platter.

Clicking on the depth text to highlight it, or clicking again to put the cursor in the text field has no reliable relationship to editing text in that text box. I just set the N7 text .svg to .4mm deep fine, then trying to set the wedge .svg to the same keeps deleting the object. I gave up, saved the project, closed Prusaslicer. Reopened the project & was promptly able to edit the depth to .4mm by just editing the dept text field. I'm unclear why it work sometimes but not others. This morning I just tried editing multiple depth attributes on the same project that was driving me nuts last night & ran into no issues. I tried clicking all over or moving the mouse away from the text box & it still entered correctly. Last night I tried repeatedly clicking on the text box, hovering the mouse exactly over the text box & other things & it would still not edit text in the text box.

I've ran into similar issues with editing other attribute numbers previously. Selecting numbers & editing them frequently seems to put input into doing things in the slicer & not into editing the text box you have text highlighted in. It doesn't seem to be consistent on where the mouse cursor is, what text is or is not highlighted. I had similar pain trying to use the add part > text function somewhat recently as well where I couldn't reliably enter text into the text box.

Screenshot 2025-03-07 220320.png shows me failing to edit the N7 red wedge item to a .4mm depth last night
Screenshot 2025-03-08 121537.png show me successfully editing the N7 text to a .4mm depth this morning
No obvious differences, except the text won't edit  on one screenshot & if you hit the delete key the part gets deleted instead of the text.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11

### Printer model

XL 5 head 6.2.2+8853 (upgraded from previous recent version lastnight)

Sage-King, 2025-03-09:
Do you have a consistent way to reproduce this issue? It is challenging to fix a bug that cannot be reproduced. Perhaps a test svg with steps on how to make it happen for us?

Phallon, 2025-03-10:
No reliable way to reproduce it yet, but I'll keep looking. I know it's happened to me at least 3 times over maybe 6 to a dozen uses of the slicer. Extremely annoying when it happens. At least I finally figured out  it appears to fix itself by restarting the slicer. I was previously thinking it was just poor UI behavior with focus following the mouse cursor not the highlighted, but now it's looking like a bug.

Jony01, 2025-03-10:
By the way, you have same issue in .svg file.
Hover mouse over this symbol to find more.
I strongly reccomend you to fix SVG file first.
[image]
[image]

Jony01, 2025-03-10:
Behavior what you describe sounds like: GUI missinterpretation of the current focus.
I will be glad for step by step way to reproduce it.
In meanwhile time you could fix it by switch focus out of(e.g. click on selected object in scene) and back into input.

Phallon, 2025-04-27:
Still no luck in reproducing it. It definitely felt like focus confusion of some sort.

closed 2026-09-29 by prusaslicer-bot[bot]