prusa3d/PrusaSlicer #14251: 3mf file parsing error

tikonen, 2025-03-09:
### Description of the bug

A 3mf export from Fusion 360 fails to parse. 

[image]

### Project file & How to reproduce

Unzip following file and open it in PrusaSlicer. The parsing fails with an error on line 10609.

[lightpen v2.zip](https://github.com/user-attachments/files/19148772/lightpen.v2.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11 24H2

### Printer model

MKS3+

Jony01, 2025-03-10:
[image]

By [3mf CORE specification](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#413-vertices) verices MUST contain at least 3x vertex:
[image]

tikonen, 2025-03-11:
I confirmed that the original Fusion model contains an empty object that when removed, the model export results again a valid 3MF. 

This looks like a bug in Autodesk Fusion export. I don't know if it makes sense for PrusaSlicer to ignore or prompt a warning for empty objects even when it's technically against 3MF spec, but in any case I consider this case closed.

closed 2025-03-11 by tikonen