prusa3d/PrusaSlicer #14252: [Feature Request] Tree (SLA) supports for FDM

Gregor-Gregor, 2025-03-09:
**A CASE:**
A model I have to print has relatively small footprint, it's growing to the side, so with every layer the mass center is moving beyond the base polygon. Imagine number seven ("7") shape. For many reasons I can't rotate it. Simply speaking, the printout will overturn, I need to support it. For FDM tgere are several choices of supports to choose.

_However._ Side walls of the object are steep, so organic/standard supports leave visible scars on the surface, difficult to remove. They prove best for horizontal or slightly raising surfaces.

**A SOLUTION:**
I believe tree (scaffolding) support might work better with steep overhang (or even slim vertical objects – imagine a launch tower for a rocket). With pinpoint tips, the contact residue will be easy to ckean up.

Gregor-Gregor, 2025-03-10:
Ahoy there!
Thanks, it works well indeed. :)

_Meanwhile, thinking about it a bit deeper:_
Perhaps it would be beneficial to add an option to declare an object (or a part) as a support?
Similar way the _Add Part –> Load_ context menu behaves.
However, a geometry added this way would be processed by the slicer the same way internally generated supports are: deditaced speed/material, but also possibly single/double wall or no infill (like organic ones). This would allow people to add their self-generated (externally generated) supports when needed.

From my own experience, many times before I was not happy with supports I was able to generate in the slicer, or needed some additional geometry to stabilize printout. Such additional piece was discarded afterwards, as the supports are, but out of necessity they were printed as part of the whole, with the same infill/perimeter/brim/whatever printing settings.

If you find this idea appealing, I can issue a separate _Feature Request_.

neophyl, 2025-03-10:
That has already been suggested before.  https://github.com/prusa3d/PrusaSlicer/issues/3390

Gregor-Gregor, 2025-03-11:
Thanks, I was looking for this 😊

closed 2025-03-12 by Jan-Soustruznik