prusa3d/PrusaSlicer #14253: Enable wipe tower on non solulable multimaterial prints

tsamisa, 2025-03-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

tsamisa, 2025-03-10:
First of all i apologize if this is requested before. The wipe tower is disabled if i user support with a materials that is not solulable and has a top contact other than 0. There sometimes the need to print with a support  material that is not easily detachable from the main print so we need a z contact > 0. Or we want to print with a .15 layer height and there is no solulable option. Please allow for the wipe tower to be enabled. Add a warning or an experimental tag similar with other options.

Braintoe, 2025-03-13:
While the Prusaslicer Wipe Tower leaves a lot to be desired (not always the same start position for each layer for example since I keep getting ripped-off or broken wipe towers with my slightly leaky extruder...) it is possible to use it as you like. As soon as I use a printer setting with more than one extruder, Prusaslicer does not care about the "soluble" setting here and offers the wipe tower to me.

tsamisa, 2025-03-14:
It offers the wipe tower but for example if i want to use petg with petg supports from another extruder, and its not a soluble interface i get a message this message : 
"The Wipe Tower currently supports the non-soluble supports only if they are
printed with the current extruder without triggering a tool change. (both
support_material_extruder and support_material_interface_extruder need to be
set to 0)."
There is prusaslicer fork that does this but i dont see the reason why not incorporating it to the official release.

Braintoe, 2025-03-14:
I do frequently print using PETG for the part on one extruder and PLA for supports on the other one, both of which are not marked as "soluble", and I get no errors at all.

Some settings that might have an effect:

-  printer is set with two extruders and relative distances for extrusion since at least Prusaslivcer 2.6 only supported those with a wipe tower.
- filament: "soluble" is set deactivated in all cases. Miniomal amount in purge tower is set to a useful value
- print settings: perimeter, infill and massive infill set to extruder 1, support material and support interface material extruder set to 2, purge tower extruder set to 0. I faintly remember setting that one to another value caused some issues. Purge tower activated.

Hope that helps, otrherwise we need to compare some more settings ;-)

tsamisa, 2025-03-15:
First of all thank you for the time to respond. 
If i use PLA as PETG support and vice versa it doesn't matter if the Z distance is 0 since they detach easily. If i use PETG with PETG supports but from lets say an older spool that i want to get rid of, the Z distance is important in order to detach., So if i allow soluble and assign let say all the infill and perimeters to Extruder 1 and supports to 2 everything is ok. Now if i go to "support material" the top contact z distance is 0. If i try to change this i get the error i showed above. My printer is set to XL 5T. SImilarly if lets say i want to print with a .15 layer height and try to assign to supports a specific extruder i cannot enable the wipe tower and i get the same message

Braintoe, 2025-03-16:
Okay, our use cases seem quite similar :-)

I tried this, and you are right, but with one unlogical exception:

- I start with the setting above but put two different PETG filaments in both extruders
- my default support style is "organic". If leave that setting, I can perfectly do what you want...
- ... but as soon as I switch it to one of the other settings I get the error message you mentioned.

That explains why I did not notice this.

tsamisa, 2025-03-16:
As i said there is fork of prusa slicer that disables this warning. I just don't get why prusa doesn't include this. If there are other issues they should allow it with an "experimental " tag or something.

JGeiger2024, 2025-06-09:
Anyone know whether an upcoming slicer version will incorporate fixes for mixed material wipe tower and support material?

closed 2026-09-29 by prusaslicer-bot[bot]