prusa3d/PrusaSlicer #14254: Flimsy thin walls from missing infill.

richfelker, 2025-03-10:
### Description of the bug

Under conditions that depend on how perimeter and infill line widths interact with the width of thin part, PrusaSlicer can end up leaving a large but thin infill region completely empty. I encountered this in the real world printing Clockspring's Phone Lounge (paid model) with 0.42 mm perimeters and 0.5 mm infill line width, in a region with a nominally 2 mm wall that turned into two separate, flimsy ~0.8 mm walls with a gap between them.

What I expected was that the gap too thin to admit infill would be replaced by solid fill or by increasing the inner perimeter width to fill the gap up to the opposing inner perimeter.

### Project file & How to reproduce

I've reproduced the issue in a minimal test case. Unfortunately the line widths needed to get it to happen here were a little more extreme in how they vary (0.45 perimeter vs 0.7 infill), but still in the realm of plausible:

[missing-fill.zip](https://github.com/user-attachments/files/19165273/missing-fill.zip)

[image]

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Linux

### Printer model

Default FFF

u89djt, 2025-03-11:
(From a fellow user) You told it not to go thin enough to fill the void.
[image]

richfelker, 2025-03-11:
As a user, what I told it is that I want to use thicker infill lines because they generally produce a stronger part and take less time to print. It may be that PrusaSlicer is interpreting that as what you said, but that's not intuitive and not obvious under what conditions a gap will be present. For example if I tune the infill line width down slightly but still large enough not to fit, the inner perimeters end up getting thickened to fill the gap. I'm not sure exactly what the mechanism going on here is, but because of this I expect it's unintended.

In the real example I hit this, I was using 0.5 mm infill line width but a perimeter line width closer to the nozzle size to get better surface detail. I generally try to avoid infill line width less than 0.5 mm because I find it contributes very little to compressive part strength and serves as little more than internal support. But because of this issue, it looks unsafe to use wider infill lines by default unless you're going to check that they didn't cause any gaps like this to get left.

richfelker, 2026-09-02:
This issue is still present. Pinging to quiet the stalebot.