prusa3d/PrusaSlicer #14258: inconsident slicing relating to nozzle size

brettowe, 2025-03-10:
### Description of the bug

noticed an issue with a model

[image]

with 0.6 nozzle
[image]

0.4
[image]

0.25 is fine tho

[image]

printer profiles used for this example is the stock mini/mini+
0.4 without input shaping also shows a minor defect otherwise above choices are with input shaper choices

zip has the step file
[loupe-jig.zip](https://github.com/user-attachments/files/19170661/loupe-jig.zip)

### Project file & How to reproduce

import step file
choose one of the prusa mini printer profiles aside 0.25 nozzle input shaper
slice and look at walls

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

win 10

### Printer model

prusa mini+

u89djt, 2025-03-11:
(From a fellow user) Your model is too thin to extrude as plastic in those places so it has to stop somewhere. The slicer converts your step file into a mesh to import it. The mesh has finite resolution. The interaction between the volume described by those triangles beats against the layer height and makes the lip look slightly different thicknesses as the slicer works up the object, so the printable region is different on different layers.
Ideally I guess you'd give it something it can print with your preferred nozzle. 
That said, you can switch to the classic perimeter generator, which takes a less agile approach to complex geometries, which is in effect what you're giving it. That seems to give a more even result.
[image]

brettowe, 2025-03-11:
i'm well aware that its too thin and i expect it to not generate the full wall, this was a test jig and not a final model
but I do expect it to generate the wall consistently until the thickness changes which doesn't happen until the top hence the bug report

u89djt, 2025-03-11:
Aye, the thickness of the mesh does actually change as you go up the body. It's a collection of flat triangles trying to follow a curved surface with different things happening at the top and bottom. Until step files can be processed directly, that's going to happen uless you break it down into subunits where the triangles can stay at consistent angles to each other. You can do that with this model easily enough, e.g. 
[prepped.zip](https://github.com/user-attachments/files/19186385/prepped.zip)
[image]

closed 2026-09-29 by prusaslicer-bot[bot]