prusa3d/PrusaSlicer #14262: Sequential printing arrangement does not consider support material

kForth, 2025-03-11:
### Description of the bug

The new sequential printing arrangement algorithm can place objects too close when support material is used, causing the toolhead to collide with already printed parts.

[image]

[image]

### Project file & How to reproduce

[Collision Example.zip](https://github.com/user-attachments/files/19197087/Collision.Example.zip)

Open project, auto-arrange models, then observe collision in g-code viewer (line 101451 for example).

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1 win64 / 2.9.1 linux

### Operating system

Windows 11, Debian 12

### Printer model

Prusa CORE One / custom printer with Dragon Burner v8

kForth, 2025-03-12:
Similarly, draft shields will cause collisions in sequential printing mode.

[image]

[image]

Both of these collision issues happen with the basic arrangement algorithm as well as the new model based one, it's just more obvious after the new update.

Some solutions could be a "per object" draft shield and a post-slicing collision check.

Mike8-I, 2025-04-21:
Hallo @Jan-Soustruznik,  as the supports can be exported with the model as an .stl , the geometry is available. As @kForth already mentioned,  a post slicing check and rearrange could be the solution.  This can be applied to draft shields, too. I understand that this is not easy, but it is part of doing it properly. So please go the extra miles and make this great feature  fit for practical use. At the moment its somewhere between alpha and  beta status.

closed 2026-09-29 by prusaslicer-bot[bot]