prusa3d/PrusaSlicer #14265: Please add Fundamental Filament Type "PCTG"

bryn51, 2025-03-12:
**Is your feature request related to a problem? Please describe.**
I would like to distinguish PCTG filament from other Types in Filament Profile

**Describe the solution you'd like**
Add PCTG to dropdown "Filament Type" on Filament Profile > Advanced screen
Also add it as a Type to Mini, MK4S, Mk4, CoreOne and XL printer firmware to match gcode.

**Describe how it would work**
This is just like other filament types.

**Describe alternatives you've considered**
PCTG filament appears in Configuration Assistant as a "Type" and when a PCTG filament is selected it shows in the Filament>advanced>Type field, but its not actually a member of the list. 

**Additional context**
PCTG material is available from Spectrum, Essentium and other filament manufacturers. It is gradually taking hold in the market,. It ought not to be ignored by Prusa Slicer.

flintobrien, 2025-03-20:
### Problem
I have custom PCTG filament profile in PrusaSlicer that is based on PETG, so the "Filament type" field is still "PETG". From PrusaSlicer, when I send a print to the printer it says I have the wrong filament type according the the g-code. The printer is behaving correctly. I added a @PCTG custom type to my MK4S. I did that because PCTG hotend temp is hotter than PETG. In order to load the filament in the printer, I need to specify Custom or (my saved @PCTG) to get the correct hotend temperature.

### Solution
Specify the same custom named filament type in the slicer as the one loaded in the printer. Then the printer prints without first going through a filament type override procedure.

bryn51, 2025-03-20:
I acknowledge that Prusa have recently added the Custom/user defined filament types in the printer firmwares.  However presently has no matching way to add user defined filament types in Prusa Slicer, hence this request.

bryn51, 2025-03-20:
Damn, you're right.  So much unexpected functionality.

bryn51, 2025-03-20:
So, maybe this request is fully catered for with that revelation.  If so, happy to cancel it, one less thing to consume scarce programming resources.

neophyl, 2025-03-20:
Ive been using it on my Klipper based machine to set different Z offsets based on the material type.

flintobrien, 2025-03-21:
Thanks. I didn't know the "Filament type" field was an editable text box. I always hit the drop-down arrow.

hansratzinger, 2025-05-24:
@neophyl Thx for this hint. Works properly even combined with Prusa Core One firmware.

closed 2026-09-29 by prusaslicer-bot[bot]