prusa3d/PrusaSlicer #14268: Multiple Authentication Bugs On Windows 11

johnelle, 2025-03-12:
One issue is that the app cannot remember the password.  I have to look it up in the password manager every time because I use strong passwords.   In addition the dialog box for logging in, is seriously buggy--the username keeps loosing focus so its almost impossible to type the username.  The password does not except the normal cut/paste menu functions.  You have to use CTRL V for pasting the password from the password manager.  During my sessions I get continual warnings that the system can't communicate with my account, even when I have manually logged in.  I have tried turning off my VPN but that doesn't fix the issue.

TerPuc, 2025-03-19:
Dear @johnelle, thank you for your report. May I kindly ask you to check if the logging dialog is still loosing focus? Thank you.

johnelle, 2025-03-19:
Focus seems to be working, but it still can't use context menu cut/paste
and it doesn't fetch the password from my Google Password Manager like most
apps--I have to look it up each time I login.

The good news is that it has remained logged in for several days.  I used
to login every time I used the app

closed 2026-09-29 by prusaslicer-bot[bot]