prusa3d/PrusaSlicer #14275: Web Access in MacOS 10.15.7

Ret-Marut, 2025-03-13:
There seems to be a problem with 2.9.1 when accessing web services like Printables and Prusa Connect tabs. Printables doesn't show any images, Prusa Connect keeps on trying to connect.
It worked with 2.9.0.

worfeus, 2025-03-14:
Confirmng this. Even 2.9.1 RC1 and RC2 worked fine.

Error when lookong for applciation updates: 
Downloading PrusaSlicer version file has failed:
Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Couldn't resolve host name:
Could not resolve host: files.prusa3d.com
[Error 6]

Also configuration updates fail.

Pasting the URL in the browser works just fine and downloads the file, so its not network connectivity for the entire computer. 

Turning off Apple Private Relay under Settings > ICloud works. Somehow its not allowing PrusaSlicer to connect.

Ret-Marut, 2025-03-15:
Yep, works for me, too. Thanx!

closed 2025-03-15 by Ret-Marut