prusa3d/PrusaSlicer #14278: Scarf seam not placed at all

aknezevi-work, 2025-03-14:
### Description of the bug

The scarf joint is not placed at all for some reason. Might have something to do with my other settings, but it makes no sense that any of them cause this, as I have enabled the scarf joint placement on non-smooth perimeters.

[image]
[image]

### Project file & How to reproduce

Here is the project file:

[Scarf_Joint_Not_Placed.zip](https://github.com/user-attachments/files/19243801/Scarf_Joint_Not_Placed.zip)

Just open, check settings and slice. Scarf Joint should be added, but is not.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1

### Operating system

Edition	Windows 11 Pro Version	24H2 Installed on	‎17/‎12/‎2024 OS build	26100.3476 Serial number	R9113LJV Experience	Windows Feature Experience Pack 1000.26100.54.0

### Printer model

Original Prusa MK4 Firmware 6.2.3-alpha+8858

u89djt, 2025-03-14:
(Fellow user taking an interest) It doesn't seem to have anything to do with your settings or your object, though I've removed a helical sprue from the top of the thread.
I can't get entire perimeter scarf to work at all, so there seems to be a bug. Slightly surprised at such a big one, so my assumption right now is that I've done something silly somewhere.
If you don't set entire, but set the scarf length to slightly less than the length of the perimeter, it goes that far round.
I set artificially low speeds so it's obvious where the scarves are.

[clampedspeeds.zip](https://github.com/user-attachments/files/19248570/clampedspeeds.zip)

[image]

This didn't make any difference, but I clipped it off to rule it out.

[image]

Jan-Soustruznik, 2025-03-17:
Hello, @aknezevi-work , Thank you for feedback, I created insternal issue(SPE-2718). We will try to fix in some of next release. Have a nice day.

closed 2025-04-14 by aknezevi-work

PerplexedFox, 2025-08-08:
I'm having the same issue. Scarf seam activated, yet after slicing the seam is still shown as a single aligned z line