prusa3d/PrusaSlicer #14279: Werkt niet met Prusaslicer wel met Cura

DannyBoyke, 2025-03-14:
[image]

Fout bij het gebruik van Prusaslicer 2.9.1.
Geen probleem bij gebruik van Cura
Oorzaak?
Hoe voeg ik .3mf file toe?

u89djt, 2025-03-14:
(From a fellow user) zip the 3mf file, as requested in the submission guidance:
"Project file & How to reproduce*
Please upload a ZIP archive containing the project file used when the problem arise."

DannyBoyke, 2025-03-14:
Het is niet mogelijk om de .3mf of .obj te laden zo kan ik ook geen project file aanmaken.
Als ik de tekening aanpas in Fusion 360 is deze wel uitvoerbaar.

u89djt, 2025-03-14:
OK, post a zip of the file(s) that are giving you the trouble.

u89djt, 2025-03-14:
I have Fusion - I'm not promising anything, but I can take a look at the source if you like.

DannyBoyke, 2025-03-14:
[2 gaten asym links 2 v5 v1 (3).zip](https://github.com/user-attachments/files/19244649/2.gaten.asym.links.2.v5.v1.3.zip)

u89djt, 2025-03-14:
Mystery mostly solved. You've got a notional body with no data in your fusion file.
[image]
That leads to an object definition in the 3mf with no data
[image]
If you don't export that null object with the true object, the slicer imports it happily.
When there's an empty object - something that can't be printed - the slicer presumably enters into a loop to deal with the data, but there's no data.
That could be ignored, or the user could be alerted to a problem. Cura is ignoring it, I guess.
If you export the Fusion document as an obj, it imports with no problem. Obj says "here comes an object, here come its points, here come its facets" sequentially for each object. A 3mf says "Here's a bunch of data" and then "here's how to build the data", and you've given it null data for part of the build.
It probably seems obvious that this should be ignored, but dodgy data can hit you years later. There's philosophy to be developed when there are so many different categories of user.

DannyBoyke, 2025-03-14:
Bedankt om het probleem uit te leggen.

neophyl, 2025-03-14:
Looks like the same issue here that was closed https://github.com/prusa3d/PrusaSlicer/issues/14251

Might be nice to handle the non compliance in PS though just to save problems in future.

u89djt, 2025-03-14:
Maybe the bug submission guidance could include this link and say "click here":
https://github.com/prusa3d/PrusaSlicer/issues?q=is%3Aissue

closed 2025-03-17 by Jan-Soustruznik