prusa3d/PrusaSlicer #14280: brim in latest prusa 2.9.1

ColinWise, 2025-03-14:
i may be being daft here but it appears as when i have the default set to no brim i get a positive tick in the plater screen 
and vice versa.

u89djt, 2025-03-14:
(From a fellow user) I don't think anyone will feel justified in suggesting you're daft without a zipped 3mf file and a screenshot showing what you mean :)

ColinWise, 2025-03-14:
Hi Dave
It must of been my setup,
somehow ,
I was ticking the box for a brim with my default setup set for an outside brim and was getting no brim if unchecked the box I was getting brim.if I saved my default without a brim I always got a brim I repeated this about 6 or seven times . This was just after installing 2.9.1 . After a computer restart there's no issue .. It could of been my settings ?
But its  ok now?
  Thanks for getting back.
Sorry to of bothered you ,
I should of done a restart first.
Best regards
Colin Wise

u89djt, 2025-03-14:
No problem at all! I have no clue - if it happens again, come back and post the file. :) You can close the issue for yourself, and re-open it if things go awry again.

closed 2026-09-29 by prusaslicer-bot[bot]