prusa3d/PrusaSlicer #14281: PrusaSlicer hanging while slicing/support generation

pamstimm, 2025-03-14:
I have been using the current version 2.9.1 without any issues, until yesterday. After importing a .zip, orienting it to best printable layout, hollowing it with a 8mm drainage hole on the bottom, and then automatically generating support structures at 100%, the program hangs up in slicing process at 14%. This is constantly happening.  After trying to hit escape it will give me the "(Not Responding)"  title at the top of the page. 

Can you please make any suggestions?

Thank you,
Pam Timm

[image]

u89djt, 2025-03-14:
(From a fellow user) 
Hi there, you need to zip the 3mf file to post it. Save it just before the slicing that hangs.

TerPuc, 2025-03-19:
Dear @pamstimm, thank you for your report. May I kindly ask you to provide the 3mf file (zipped) as @u89djt is suggesting? We would like to investigate this case. Thank you.

felipebalbi, 2025-09-01:
I have the same problem with 2.9.2:

[image]

`btop` says it's taking 12.5% CPU:

[image]

But it never moves anywhere. Here's the STEP file I'm using:

[Case-Lid.zip](https://github.com/user-attachments/files/22083610/Case-Lid.zip)

FYI, I'm on Arch Linux, in case that gives a hint of what might be going on. I'm also attaching trace logs.

[prusa-slicer.log](https://github.com/user-attachments/files/22083652/prusa-slicer.log)

u89djt, 2025-10-22:
(community triage) @pamstimm @apertureinnovations you need to save the project to a 3mf, zip that 3mf file, and post that zip file. You also need to follow the issue submission instructions in terms of the specifications of the hardware and operating system you're running the slicer on. If you do that, someone will be able to try to find clues as to the problem. If you don't, there's nothing for anybody to try.
@pamstimm the step file you posted is not a problem, it slices just fine for me, just like 99.9...9% of things. You're experiencing something down in the 0.0...01 end of things and you need to provide the information specified in the submission instructions.

closed 2026-09-29 by prusaslicer-bot[bot]