prusa3d/PrusaSlicer #14284: Organic supports, no bottom interfaces generated on top surfaces of model

zjohnson16, 2025-03-14:
### Description of the bug

I'm printing PLA with PLA organic supports and PETG "soluble" support interface. Everything works perfectly, except that when the tree supports need to be generated on top of the model, bottom interfaces of PETG "soluble" support are not created on the top surfaces of the model. This causes PLA to be printed directly onto PLA, and results in scarring on the top surfaces of the model.

[image]

If bottom interfaces were generated onto the top surface of the model, the organic supports would come off easily with no scarring.

### Project file & How to reproduce

[OrganicSupport.zip](https://github.com/user-attachments/files/19253871/OrganicSupport.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1

### Operating system

Windows 11 24H2

### Printer model

Prusa MK4S

u89djt, 2025-03-14:
Bottom interfaces are deactivated in your 3mf file
[image]
Activating them doesn't help, though.

zjohnson16, 2025-03-14:
I think I can get it to work if I tune the speeds and temps. After all, the PETG interfaces adhere to the PLA supports and the PLA adheres to the PETG interfaces well enough. It's not a complete oil/water situation, there is a little bit of a bond that is formed.

But yeah, it's some sort of interaction with the algorithm that generates organic supports that prevents the bottom interfaces from being created. Although maybe the best solution would be to not have an interface at all but just ensure that the first X layers of the organic support above the top model surface are printed in the interface material.

I did think of a potential workaround that I'll try later, maybe if I add small cylinders  to the model and set them to the PETG extruder I can create my own custom "interfaces" under the organic supports that get generated. Adding the cylinders might change where the supports get created though, I'll have to test.

u89djt, 2025-03-14:
You could add a 0.15 solid layer of petg on the disc?

Jan-Soustruznik, 2025-03-17:
Closed as solved

closed 2025-03-17 by Jan-Soustruznik