prusa3d/PrusaSlicer #14286: PeusaSlicer 2.9.1 .3MF error

JHSurf, 2025-03-15:
### Description of the bug

When trying to slice a .3mf file from Printables withing Prusa Slicer or when trying to import or open a local .3mf file I receive the error message below.

Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 46

[image]

### Project file & How to reproduce

Unzip file
Open the .3mf file

[elegoo-centauri-carbon-low-riser-remix-model_files.zip](https://github.com/user-attachments/files/19261642/elegoo-centauri-carbon-low-riser-remix-model_files.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1

### Operating system

Windows 11

### Printer model

Prusa Mk3s

neophyl, 2025-03-16:
Its a Bambu Labs created 3mf.  The changes they have made to their version of the slicer makes most of the files no longer interchangeable which is a pity.

8BitSaint, 2025-03-17:
> Its a Bambu Labs created 3mf. The changes they have made to their version of the slicer makes most of the files no longer interchangeable which is a pity.

I'm having the same issue with 2.9.1 (I don't know if it was an issue before 2.9.1) and an OrcaSlicer generated 3mf. In OrcaSlicer, I tried `File > Export > Export Generic 3MF...` and still get the same error.

[image]

From my understanding, [`Export Generic 3MF...` was introduced to work around incompatibility issues](https://github.com/bambulab/BambuStudio/issues/169#issuecomment-1766271654) between BambuStudio generated 3MF files and PrusaSlicer. Additionally, [this PrusaSlicer PR that was merged almost exactly a year ago](https://github.com/prusa3d/PrusaSlicer/pull/10808) supposedly addressed the incompatibilities as well, perhaps there was some regression?

Related or not, I only get the error when the 3MF has a piece with a modifier on it (generic, support enforcer/blocker, negative part). I don't get the error when the 3MF only has parts with no modifiers, or if the only "modifier" is an added part.

Jony01, 2025-03-17:
`elegoo-centauri-carbon-low-riser-remix.3mf` contain on line `46` include of object `id=12` from file `../object_5.model` object type is set to `other`. 
[image]
By 3mf specification[[link](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#41-meshes)]: 

> Objects of type "other" MUST NOT be referenced in the Build section, either recursively or directly.

It is up to 3mf creator to follow specification.

closed 2026-09-29 by prusaslicer-bot[bot]