prusa3d/PrusaSlicer #14288: option de sortie

Couski61, 2025-03-15:
Bonjour , j ai la dernière version de prusaslicer ,et quand je vais dans ,,,,"Option de sortie ", je ne trouve plus les paramètres pour changer la hauteur et la distance entre les pièces ( radius and height) .j ai une creality s10 smartpro
merci
Stéphane

neophyl, 2025-03-16:
Those settings are now in Printers>General and are Red/Expert level settings.

[image]

The change was mentioned in the release notes.  https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1
Specifically this line "The values were moved from Print Settings to Printer Settings, as they describe the physical geometry of the printer."

Couski61, 2025-03-16:
Rebonjour ,merci pour votre réponse rapide ,,un autre chose avec cette fonction , je ne vois plus sur le plateau, le dessin "cercle" qui délimite la zone que j' ai paramétré qui facilite le placement des plusieurs objets ,est-ce normal ?  Est-il possible d' installer une version antérieure? 
merci 
Stéphane

neophyl, 2025-03-17:
Correct Prusa seem to have removed that as part of the changes.  Someone else has already raised that as an issue here https://github.com/prusa3d/PrusaSlicer/issues/14240

You can run older versions, just download them from here, unzip to different folders and then just run the .exe files (at least on Windows based systems). 
The problem occurs when a newer version updates the profiles and so is no longer compatible with the older version.  However you can get around that too by specifically defining the profile location as part of a startup batch file.  When done that way you can run as many totally separate versions as you like.  As each will have its own profile folder.

Creating a batch file with a suitable name in the same directory as the exe of the version you want to run with the following inside
prusa-slicer.exe --datadir %cd%
will cause Prusa slicer to start and create/use a profile directory located in the same sub folder when that batch file is run.