prusa3d/PrusaSlicer #14297: Make autofill work in PrusaSlicer's Prusa Account login

garrettreid, 2025-03-17:
**Is your feature request related to a problem? Please describe.**
When logging into my Prusa Connect account in PrusaSlicer on macOS, the login dialog appears within the application and doesn't autofill. As is best security practice, I have a random password for Prusa that is stored in my password manager. Without autofill, logging in involves opening the browser/password manager, looking up details, and copying them over.

**Describe the solution you'd like**
I'd like the authentication to leverage Apple's APIs for web authentication within an application, so that saved passwords can autofill.

**Describe how it would work**
I think the best way to build this would be to use an ASWebAuthenticationSession, as per the Apple Developer documentation on [Authenticating a User Through a Web Service](https://developer.apple.com/documentation/authenticationservices/authenticating-a-user-through-a-web-service).

**Describe alternatives you've considered**
Alternatively, this could also be done by opening a URL directly in the system browser, which then calls back into the app with a custom URL scheme. This isn't quite as seamless. If you've gone through authentication in Fusion 360, I think that's how they implemented it - you can see the browser pop up then hand back over to the app.

**Additional context**
I'm not sure how Windows or Linux password management works, but I suspect users on those platforms would also appreciate the same courtesy :)

rob-miller, 2025-04-07:
On 15.3.2 (Sequoia) I get the Autofill menu when clicking in the teztfield, and can fill the password from the system database.  

Unfortunately #12936 I am unable to paste into the password field so this is only helpful for newer systems.

garrettreid, 2026-09-02:
Confirmed this now works natively in PrusaSlicer 3.0.0-alpha11 - hooray!
Before that, rob-miller's workaround is functional, if a little slow.

closed 2026-09-02 by garrettreid