prusa3d/PrusaSlicer #14300: XL 5T: Changing extruder print order results in one tool not being used

alfrankgit, 2025-03-17:
I have a box on extruder 1 and a logo on extruder 2.

The default is to print with extruder 1 first and then with extruder 2. See first image: the logo gets printed.

[image]

But I want the logo on extruder 2 to be printed first, because it's small and only on the first layer and  might be partly blocked by the existing lines of extruder 1.

To achieve this I drag the logo on extruder 2 in the list of the object names one line up, so that it now is above the box.

But now the logo is gone in the preview and it doesn't get printed anymore. See the second image.

[image]

Is this a bug or am I approaching this the wrong way?

Best regards,

Alex

m29corey, 2025-03-17:
I'm guessing that there is not a void for the logo in the body.

It looks like for a single object with multiple parts, later parts take precedence if there is geometry overlap, and without a void for the logo, the body takes precedence.

But it gets a bit weirder.

If you put the logo back under the body in the list, but change the extruder of the body to a higher number extruder than the logo, then the logo will be extruded first.

alfrankgit, 2025-03-18:
No, there's no void in the body for the logo.
I added the logo to the body via the context menu "Add part / load".
And this makes space in the body as long as the body gets printed first.

Yes, I already thought about putting the body on the second extruder and the logo on the first.
But this seems an odd thing to do.

Especially since changing the printing order of the tools or filaments works perfectly on Orca Slicer or Bambu Studio via the plate preferences.

closed 2026-09-29 by prusaslicer-bot[bot]