prusa3d/PrusaSlicer #14301: Profile Inheritance

nomike, 2025-03-17:
**Is your feature request related to a problem? Please describe.**

Printing-, printer- and filament-profiles are being updated every now and then. Settings are fine tuned, etc..

I do have some customizations though. For example:

- Prices for Filaments: as they vary
- Custom End-G-Code: I added a few custom M300 commands to play a tune once the heat-bed has cooled down enough at the end of the print.
- Infill: I prefer adaptive cubic infill on my prints per default

For that I need to create custom profiles though and will miss out on any improvements done by Prusa when the built-in profiles receive updates.

**Describe the solution you'd like**

I would like to see an inheritance feature implemented for profiles. Each profile would need another profile as a parent. I could then select specific properties to be overwritten, with the rest just being inherited from the parent.

Multiple levels of inheritance could be done. Which would also make existing profiles be simpler and more streamlined. The "Prusament PLA" profile could for example be based on "Generic PLA", as most of the settings would be the same.

**Describe how it would work**

Each profile would have a new mandatory field named "parent" which is rendered as a DropDown GUI element. It allows the user to select any of the available profiles.

Options within the profile would be presented as usual, but each of them would have an additional checkbox, oriented to the right of the screen (at least on left-to-right languages). GUI elements for changing settings would be disabled and revert to the parent setting when the corresponding additional checkbox is unckecked. When it is checked, the GUI element would be enabled and the user would be free to change the value (or keep it the same, to "freeze" the setting to it's current value even though it might be changed in the parent profile later).

This way the profile would only contain settings I actively want to change. If for example I change the price of a roll of "Extrudr PLA NX2" filament, I will still get updated fan-speed settings if Prusa optimizes them.

Settings currently are grouped in the GUI. As an optional convenience feature for users, there could be an additional checkmark on top of each group, which allows the user to check or uncheck all boxes within that group.

When changing the value of "Parent" (in the "changed" signal of the DropDown element), a quick check should be performed to detect circular dependencies (Profile A depending on Profile B which in turn depends on Profile A) and a warning should be displayed to the user and the DropDown reverted to it's previous value.
Alternatively these checks could be performed beforehand to not even include invalid profiles in the list or maybe even better still include them but have them disabled, so that users see that they do exist, but know that they can't select them.

**An example**

I'm professionally administrating a few F5 BigIP Load Balancers and got this inspiration from their Web-Interface, as this is exactly how they do it.

Here's a screenshot to visualize this:

[image]

**Describe alternatives you've considered**

There aren't really any alternatives to achieve a similar output.

Currently PrusaSlicer has the option to compare profiles. So in theory I could open the "Extrudr PLA NX2" profile, save it under a different name ("My Extrudr PLA NX2") with the updated price, and diff it to the original every time I've installed a configuration update, to manually migrate over any potential changes.
This would be very tedious.

Having multi-level-inheritance (e.g. basing my profile on "Extrudr PLA NX2" which in turn is based on "Generic PLA") would be exponentially more tricky.

**Additional context**

I believe that this will not only benefit the users, but will also make it much easier for Prusa to manage all the profiles. With more and more filaments and printers being added, things will inevitably get increasingly convoluted.

Also it looks like this feature is already partly implemented as can for example be seen with those lock icons. So we're already half-way there.

[image]

neophyl, 2025-03-17:
That's how it works already.  They do have multi level inheritance.  One PLA will have another PLA as a base. etc.  The profile on top of that only has the changed settings in the profile, the rest it uses from the 'Parent' profile.  Which can in itself be based on another etc.  So a change to the underlying profile will be reflected in the later 'child' one UNLESS that profile specifically has its own value for a setting.  So you can end up with a big chain of profiles.

It is already incredibly convoluted.

You have to use the Detach From System Preset button to break that inheritance if you don't want ANY Prusa changes to make their way into your carefully tuned profiles.

The one thing there isn't is some form of management system for this.  Something that was asked for a long time ago.

castaway, 2026-09-02:
Still could do with improvements - drives me nuts if I ever close the Slicer ..