prusa3d/PrusaSlicer #14304: Collision on auto-generated sequential printing

dseight, 2025-03-17:
### Description of the bug

The front fan has collided with a relatively tall object during the sequential printing. The layout/order of print objects was auto-generated.

### Project file & How to reproduce

The order of items was the following:
[image]

The collision occurred when moving from item 3 to item 4. The moment can be seen on the following screenshot:
[image]

Project file and exported bgcode on which the issue occured: [collision.zip](https://github.com/user-attachments/files/19297894/collision.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.1

### Operating system

Windows 11

### Printer model

Prusa MK4S

Jan-Soustruznik, 2025-03-19:
Hello, @dseight, thank you for sharing your case. I have created an internal issue (SPE-2721). We will try to fix it in one of the upcoming releases. Have a nice day!

RajHid, 2025-03-23:
had the same issue.
Also the slicer mentioned a colision. Checking it in the Slicer preview wase fine.
==> Something is off! 
Is there a way to warn people from using certain features if there are known bugs aparent?
--> The Slicer knows if there are updates so there is a channel to provide newest informations.

thew1se, 2025-03-30:
I also had issues.  5 parts all identical, and they were auto-placed.  Order of print was back right, to front left.  This is the opposite of how it should be.  Should be left to right, front to back, shortest to tallest.

[image]

dseight, 2025-05-24:
I can confirm that it works correctly with 2.9.2. I guess, this can be closed then?