prusa3d/PrusaSlicer #14305: Align models feature

abhishekgyani, 2025-03-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Hey, i want a feature where i can align the models with each other, align either to center or to one face etc

**Describe how it would work**
Be as descriptive as your skill level allows you.

I upload two models , and I want to align models to the center, or to the face of the models, just we use align in tinkercad.

u89djt, 2025-03-18:
An extremely common request. I think you can be confident that they've decided where it is in their priorities. https://github.com/prusa3d/PrusaSlicer/issues/1425

Tupson444, 2026-09-19:
still relevant