prusa3d/PrusaSlicer #14307: Top perimeters marked as overhang perimeters

petaflot, 2025-03-18:
### Description of the bug

Part has perimeters falsely marked as overhangs

may be related to https://github.com/prusa3d/PrusaSlicer/issues/14306

![Image](https://private-user-images.githubusercontent.com/32837871/423953999-bdf43c6a-d9e7-41bc-b579-044b463170a5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDIyOTgyMzYsIm5iZiI6MTc0MjI5NzkzNiwicGF0aCI6Ii8zMjgzNzg3MS80MjM5NTM5OTktYmRmNDNjNmEtZDllNy00MWJjLWI1NzktMDQ0YjQ2MzE3MGE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAzMTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMzE4VDExMzg1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNlZGY4Y2M4MTM3NWY2MjY0MTI5M2E5MmQ4NzNlOWM3MDY3YzI3NWMyYzA0YTVjOTU4MDY1ZTBlMmU4NzFiMGImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.dEF_WNyqDK7tTcOFPtLbJtrsaFkIhwWMbmUUyDKwSbw)

also see https://github.com/user-attachments/files/19317571/pawn2.zip

### Project file & How to reproduce

I tried shutting down the computer, running a memcheck... always the same.

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.8.1

### Operating system

Gentoo default/linux/amd64/23.0/desktop (stable)

### Printer model

custom build derived from CR10s Pro running Marlin

u89djt, 2025-03-18:
(From a fellow user) They are overhangs. The combination of angle of slope and layer height means that it's impossible to extrude a contiguous surface with any overlap, so those extrusions are unsupported. You can close the issue for yourself.
[image]

petaflot, 2025-03-18:
I see. In this regard, they indeed are overhangs. Closing, because this obviously relates to #14306 (there should be 2 shells)

closed 2025-03-18 by petaflot

u89djt, 2025-03-18:
It's too thin for two shells.
[image]