prusa3d/PrusaSlicer #14309: new version 2.9.1 make error

Feika1981, 2025-03-18:
### Description of the bug

`home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:196:5: error: ‘png_image’ was not declared in this scope; did you mean ‘png_time’?
  196 |     png_image image;
      |     ^~~~~~~~~
      |     png_time
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:197:13: error: ‘image’ was not declared in this scope; did you mean ‘Image’?
  197 |     memset(&image, 0, sizeof(image));
      |             ^~~~~
      |             Image
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:198:21: error: ‘PNG_IMAGE_VERSION’ was not declared in this scope
  198 |     image.version = PNG_IMAGE_VERSION;
      |                     ^~~~~~~~~~~~~~~~~
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:200:10: error: ‘png_image_begin_read_from_memory’ was not declared in this scope
  200 |     if (!png_image_begin_read_from_memory(&image, png_data.data(), png_data.size()))
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:203:20: error: ‘PNG_FORMAT_RGBA’ was not declared in this scope
  203 |     image.format = PNG_FORMAT_RGBA;
      |                    ^~~~~~~~~~~~~~~
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:206:23: error: ‘PNG_IMAGE_SIZE’ was not declared in this scope
  206 |     image_data.resize(PNG_IMAGE_SIZE(image));
      |                       ^~~~~~~~~~~~~~
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:207:10: error: ‘png_image_finish_read’ was not declared in this scope
  207 |     if (!png_image_finish_read(&image, nullptr, image_data.data(), 0, nullptr)) {
      |          ^~~~~~~~~~~~~~~~~~~~~
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:208:9: error: ‘png_image_free’ was not declared in this scope
  208 |         png_image_free(&image);
      |         ^~~~~~~~~~~~~~
/home/feika/Downloads/PrusaSlicer-version_2.9.1/src/libslic3r/PNGReadWrite.cpp:215:5: error: ‘png_image_free’ was not declared in this scope
  215 |     png_image_free(&image);
      |     ^~~~~~~~~~~~~~
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:1994: src/libslic3r/CMakeFiles/libslic3r.dir/PNGReadWrite.cpp.o] Ошибка 1

### Project file & How to reproduce

All versions were built normally, but in the last one this error is always present.``

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Kubuntu 22.04

### Printer model

Ellegoo

Feika1981, 2025-03-22:
Building dependencies error

-- Downloading...
   dst='/home/feika/Downloads/PrusaSlicer-version_2.9.1-rc2/deps/build/downloads/MPFR/mpfr-4.2.1.tar.bz2'
   timeout='none'
   inactivity timeout='none'
-- Using src='https://www.mpfr.org/mpfr-current/mpfr-4.2.1.tar.bz2'
CMake Error at dep_MPFR-stamp/download-dep_MPFR.cmake:163 (message):
  Each download failed!

    error: downloading 'https://www.mpfr.org/mpfr-current/mpfr-4.2.1.tar.bz2' failed
          status_code: 22
          status_string: "HTTP response code said error"
          log:
          --- LOG BEGIN ---
          timeout on name lookup is not supported

I check wget https://www.mpfr.org/mpfr-current/mpfr-4.2.1.tar.bz2

Have you even tried assembling the assembly yourself?

themanyone, 2025-03-23:
Downloading dependencies "worked for me", but I had to enable VPN to bypass blocked sites.

lukasmatena, 2025-04-07:
@Feika1981 The issue with downloading MPFR is fixed as of 2.9.2-rc1.

Feika1981, 2025-04-13:
> Downloading dependencies "worked for me", but I had to enable VPN to bypass blocked sites.

Is there any way to find out which sites are blocked? Version 2.9.2 is out and I'm still stuck in this error.

closed 2026-09-29 by prusaslicer-bot[bot]