prusa3d/PrusaSlicer #14312: Enable Scarf start height to have mm as well as % option

aknezevi-work, 2025-03-19:
**Is your feature request related to a problem? Please describe.**

Using scarf seams, it can get tricky because the printer will try to print very low height. Scarf seam start height should eliminate the problem of too low of print height, but when using variable layer height the lower layer height can further reduce the start height, making it unstable again, while at higher heights it increases the start height, reducing effectiveness of scarf seam. It would be great if a mm value instead of a % value could be set.

**Describe the solution you'd like**

When inputing Print Settings -> Layers and perimeters -> Advanced -> Scarf start height, there should be an option for mm or %, similar to Seam gap distance.

**Describe how it would work**

Well, it would read if there is a percent (%) sign in the input and use that, otherwise use mm. Similar to Seam gap distance, or Quality (slower slicing) -> Avoid crossing perimeters - Max detour length.

**Describe alternatives you've considered**

There are really no other alternatives I could think of, aside from accepting current way or using fixed layer heights.

**Additional context**

SachCZ, 2025-03-20:
Hello @aknezevi-work, we have kept the value just in `%` deliberately, to even enable scarf seams with variable print height in the first place. This is us looking at the bigger picture, where there are multiple user experience issues with the value in mm. Our user base is broad and not everyone is an expert such as you. The value in mm is very easily prone to misconfiguration.

1. In the most basic case you would be able to set the value to anything, which can lead to extrusions starting higher then current layer height.
2. Based on the first objection it would be reasonable from us to not let the user input a value bigger than current layer height, but this is where the adaptive layer height comes in which can again lead to invalid extrusions. This mistake is made very easily.
3. This means we would probably need to cut off the value on each layer

The only viable option I see is nr. 3. and it is still pretty obscure as the result might be very different to what the user expects. Can you please give as your thoughts on this UX side of things?

If we see there is a greater demand for this kind of thing, we will try to implement it. In any case, thank you for your suggestion.

aknezevi-work, 2025-03-20:
Thank You @SachCZ for the response.

Regarding the objection, several solutions are possible:

1. Just ignore the basic users? With all the respect to newcomers, but there are 3 different levels of complexity, and new user using Expert mode probably should be ignored. It is impossible to take into consideration everything the user can do, and Slicer is by default in Beginner mode, the user has to want to tinker to change all these settings. Besides, there are many settings which specifically state that it is up to user to take care (For example, not Expert, but Intermediate setting Print Settings -> Multiple Extruders -> Wipe Tower -> No Sparse Layers (EXPERIMENTAL)).
2. The default setting is 0 anyway, which does not make a difference if it is 0% or 0.00 mm. It only makes a difference if the user explicitly wants to increase the value, have a pop-up, like if Extrusion Ratio is changed to 999?
3. Have two input boxes, one for % and one for mm, and if mm value is too large (larger or equal to the current layer height) use the percentage value?
4. Make sure that, if value is in mm, it cannot be larger than the value in Printers -> Extruder X -> Layer Height Limits -> Min? I know that this one is probably the most difficult one to code, but others may not be very difficult compared to the current way.

Again, thank You for the response.

SachCZ, 2025-03-20:
Hi, thank you for your suggestions. I am actually leaning towards something along the lines of the 4. suggestion. We will see what we can do. To be completely transparent with you, it is not a high priority right now, but it is for sure a valuable contribution.

closed 2026-09-29 by prusaslicer-bot[bot]