prusa3d/PrusaSlicer #14314: Text / font broken in Debian 12 VM

TheYang, 2025-03-19:
### Description of the bug

When launching PrusaSlicer in my up to date Debian 12 VM, all the main-window text is broken.
https://imgur.com/a/aSJ5www

### Project file & How to reproduce

Create a Debian 12 VM
install PrusaSlicer (old appimage or newer flatpak)
launch PrusaSlicer
see broken text

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1, 2.8.1

### Operating system

Debian 12 VM, Windows 11 Host

### Printer model

irrelevant

SachCZ, 2025-03-20:
Hi, would you mind sharing with us the steps to replicate the issue? Meaning a concrete software and where did you obtain the ubuntu image etc. It would greatly help us in troubleshooting as of right now it is very hard based on just the general steps you provided.

TheYang, 2025-03-21:
Hey there, so It's an older Debian VM, and I'm unable to reproduce this with spinning up a new VM.  

Assuming this is an issue with fonts, I've now tried comparing the fonts of the old vm and the new VM, found a handful of fonts that the new one had, but the old one didn't, so I merged the /usr/share/fonts on the older vm. No Change.
I then supposed that maybe a font was present, but the file was broken, so I deleted all fonts from the old VM, and installed the fonts from the new vm. No Change.  

Because I remember it working, I then went to check older versions. I went back until the 2.6.1 Appimage, where I noticed that I actually get fonts-errors!

`[2025-03-21 07:34:02.623535] [0x00007f634a1edc00] [trace]   'Cantarell 11' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(43), 
[2025-03-21 07:34:02.625114] [0x00007f634a1edc00] [trace]   'Sans 9' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(43), 
[2025-03-21 07:34:02.625216] [0x00007f634a1edc00] [trace]   'Serif Italic 11' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(43), 
[2025-03-21 07:34:02.625245] [0x00007f634a1edc00] [trace]   'Sans 11' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(false), IsUsingSizeInPixels(false), Encoding(43), 
[2025-03-21 07:34:02.625344] [0x00007f634a1edc00] [trace]   'Monospace Bold 10' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(true), IsUsingSizeInPixels(false), Encoding(43), 
[2025-03-21 07:34:02.637039] [0x00007f634a1edc00] [error]   Can not read font('Sans'), file path is empty.
[2025-03-21 07:34:02.639735] [0x00007f634a1edc00] [error]   Can not read font('Font is NOT ok.'), file path is empty.
[2025-03-21 07:34:02.641446] [0x00007f634a1edc00] [error]   Can not read font('Sans'), file path is empty.
[2025-03-21 07:34:02.644157] [0x00007f634a1edc00] [error]   Can not read font('Font is NOT ok.'), file path is empty.
[2025-03-21 07:34:02.645726] [0x00007f634a1edc00] [error]   Can not read font('Serif'), file path is empty.
[2025-03-21 07:34:02.653169] [0x00007f634a1edc00] [error]   Can not read font('Font is NOT ok.'), file path is empty.
[2025-03-21 07:34:02.660672] [0x00007f634a1edc00] [error]   Can not read font('Sans'), file path is empty.
[2025-03-21 07:34:02.662740] [0x00007f634a1edc00] [error]   Can not read font('Font is NOT ok.'), file path is empty.
[2025-03-21 07:34:02.666489] [0x00007f634a1edc00] [trace]   'Monospace Bold 10'font descriptor string param of load_wxFont()
[2025-03-21 07:34:02.666543] [0x00007f634a1edc00] [trace]   'Monospace Bold 10' wx string descriptor
[2025-03-21 07:34:02.666585] [0x00007f634a1edc00] [trace]   loaded font is 'Monospace'.
[2025-03-21 07:34:02.670534] [0x00007f634a1edc00] [trace]   'Monospace Bold 10' wx string get from GetNativeFontInfoDesc. wxFont IsOk(true), isNull(false)IsFixedWidth(true), IsUsingSizeInPixels(false), Encoding(43), 
`

This is with loglevel 5.

I have to agree with the errors, I do not have fonts named "Sans" or "Serif".  
Checking it, my system-font Cantarell (as named by screenfetch, no Idea where to find that elsewhere, another random find) is a Sans font, so doesn't have Sans or Serif variations. Same for the working, new vm.

closed 2026-09-29 by prusaslicer-bot[bot]