prusa3d/PrusaSlicer #14325: Colorize the X, Y, and Z entrys

HScottBuck, 2025-03-20:
Minor feature requests:

1) In "Part manipulation" / "Group manipulation" panel, color the X Red, Y Green, and Z Blue to match the colored arrows in the 3d view.

2) Make the Red/Green/Blue arrows in the 3d view always visible on the screen.  (This should be optional.)
Always visible could be in a fixed corner of the view, near the edge of the view closest to where the arrow would normally be located on the build plate (at 0,0,0).

neophyl, 2025-03-20:
Point 1 already is

[image]

Your second item is effectively a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4410 although in slightly different specific form.  The functionality requested in 4410 would fulfil the requirement though.

HScottBuck, 2025-03-20:
Thanks!  Soo many options, glad this one is there!

u89djt, 2025-03-22:
This wants a solution: searching for "colour" reveals one list of colour-related settings, and searching for "color" gives a different list.

dhlavaty, 2025-03-24:
It should be ON by default imho

u89djt, 2025-03-24:
For the majority maybe, but all it does for colourblind folk is reduce contrast. I don't want it on - I've memorized the order of colours and axes, but I spend a lot of time in CAD and printing.

neophyl, 2025-03-24:
There's an often unwritten 'rule' of software development too that applies to this function.  Whenever you add a new option the new options default should be such that it does not cause an existing installation to change behaviour.  

As such the default being OFF fulfils that.  Before it was an option it was non coloured, the default keeps it non coloured so doesn't effect those who don't care/want the new option but allows those who do want it to enable it.

> It should be ON by default imho

While a case can be made that new installs should have the option ON then that's yet another switch case to add to the installer.

u89djt, 2025-03-24:
@dhlavaty The 'Y' here looks like it should fail a test? If they go for colouring just the column headers I guess there are design decisions to reach. 
[image]

dhlavaty, 2025-03-24:
> The 'Y' here looks like it should fail a test? 

Yes, these icons are bad. Blue is OK - contrast passes. Red would be OK for large texts only, but is not in this case as the text (letter "X") is small. And the green is really bad.