prusa3d/PrusaSlicer #14329: Bridges on top layers related to Commit b133579

locki-cz, 2025-03-21:
### Description of the bug

Bug after this https://github.com/prusa3d/PrusaSlicer/commit/b133579126e6cee44f841f66dbae018455e32117 from 2.7.3 alpha, in 2.7.2 it is ok ...

@lukasmatena can you please look on this?

### Project file & How to reproduce

How to reproduce it:
Any object with threaded holes (size doesn’t matter)
Layer height doesnt matter
A single perimeter on the top layer

[image]

[image]

[test-bridge-top-layer.zip](https://github.com/user-attachments/files/19396053/test-bridge-top-layer.zip)

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 10 x64

### Printer model

Voron 2.4

locki-cz, 2025-03-24:
fixed in orcaslicer: [https://github.com/SoftFever/OrcaSlicer/pull/9053/files](https://github.com/SoftFever/OrcaSlicer/pull/9053/files)

Jan-Soustruznik, 2025-04-10:
Hello, @locki-cz , could you share your project file(3MF). Thank you

locki-cz, 2025-04-10:
> Hello, [@locki-cz](https://github.com/locki-cz) , could you share your project file(3MF). Thank you

[test-thread.zip](https://github.com/user-attachments/files/19689249/test-thread.zip)

it is default profile, only with one perimeter on firts and top layer, nothing special

Jan-Soustruznik, 2025-05-06:
Hello @locki-cz, thank you for your report. I’ve created an internal issue (SPE-2762). We’ll aim to address it in one of the upcoming releases. Have a great day!

locki-cz, 2025-05-12:
Díky!

lukasmatena, 2026-09-01:
Fixed in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena